openspyrit/spyrit #12: Reconstruction with multiple channels

guilhermebene, 2021-05-25:
In `spyrit.learning.model_Had_DCAN`, the function `forward_N0_maptoimage_expe` works for reconstruction of a batch (`b != 0`) but does not work for a multichannel reconstruction (`c != 0`).
The problem comes from the following lines: 
```
N0_est = self.max(x_est);
N0_est = N0_est.view(b*c,1,1);
N0_est = N0_est.repeat(1,1,self.M)
x = torch.div(x,N0_est)
```
`N0_est` has the dimensions `(b*c, 1, self.M)` , whereas `x` has the dimensions `(b, c, self.M)`.

It seems a possible solution would be to convert `x` to the same dimensions as `N0_est` at the beginning of the function.

guilhermebene, 2021-05-25:
A similar behavior happens in `noiCompNet.forward_N0_maptoimage`.

guilhermebene, 2021-07-09:
This will be corrected in #24, by removing the combination of batches and channels in a single dimension during training and reconstruction.

nducros, 2021-07-09:
Coming back to your previous posts. `N0_est` should have the dimension `(b, c, self.M)`, just as the input `x`. Make sure one `N0_est` is computed for each batch in each channel. I haven't checked the behavior of `self.max()` ; it might be sufficient to remove `N0_est = N0_est.view(b*c,1,1);` ?