openspyrit/spyrit #29: pip install spyrit fails on CREATIS cluster (Fedora)

cmouton, 2021-07-21:
I followed Thomas's recommendation : no more using the cache. 
But it is still downloading the .tar.gz and failing.

Is it because there is no wheel for Fedora?

[image]

cmouton, 2021-07-21:
I can precise the Python version :  
(test_DLMIS) [mouton@linux1 openspyrit]$ python --version
Python 3.9.6

I understand that it is not supported yet (https://pypi.org/project/spyrit/#files).

cmouton, 2021-07-22:
@tbaudier : Have you already planned to add the wheels also for Python 3.9?

tbaudier, 2021-07-22:
https://github.com/openspyrit/spyrit/pull/32

tbaudier, 2021-07-23:
With python 3.9 and windows, I need to update numpy because v1.19.3 is not available for this version. But for numpy > 1.20.X there is no wheel for python 3.6. I need to fix the version of numpy because fht is an extension of numpy. So I have different possibilities:
 - Do not update numpy and do not create the wheel for windows and python 3.9
 - Update numpy and do not support python 3.6 anymore

nducros, 2021-07-23:
What would be a third option without fht that we no longer need?

tbaudier, 2021-07-23:
We can remove the fixing of the numpy version and it should work for different version of python

nducros, 2021-07-23:
Great. So this is the right time to discard it! 

You should only need to replace this [line ](https://github.com/openspyrit/spyrit/blob/36c90d10e9ca4d82671d24d68db8257a4b8e1c4d/spyrit/learning/model_Had_DCAN.py#L561) with 
`H = wh.walsh2_matrix(n)/n`

Both `Stat_h` and `optim_had `are no longer necessary as I reimplemented the walsh version of both functions (already included)

closed 2021-09-08 by tbaudier