openspyrit/spyrit #30: pip install spyrit fails on MacOS (High Sierra)

cmouton, 2021-07-21:
I followed Thomas's recommendation : no more using the cache.
But it is still downloading the .tar.gz and failing.

Is it because there is no wheel for MacOS (High Sierra)?
[image]

tbaudier, 2021-07-21:
Can you tell us the python version please ?

The wheel are available for python 3.6 -> 3.8 (https://pypi.org/project/spyrit/#files). I can add python 3.9 now. But for older python, spyrit will not work

cmouton, 2021-07-21:
Here is the version : 
(2021_DLMIS_Hands-on_test) (base) ei-cm-566:virtualenv clairemouton$ python --version
Python 3.7.3

Thanks for the link.
I saw that MacOSX is supported for 10.14, but mine is 10.13.6... That should be the reason of the failure.

closed 2021-07-21 by cmouton