openspyrit/spyrit #31: Method `forward_Pinv` is not defined in compNet

guilhermebene, 2021-07-21:
The method `forward_Pinv()` is not defined in compNet, but is called by `evaluate_Pinv()`. As seen in:
https://github.com/openspyrit/spyrit/blob/0ff170fc46c5848738e70d28b3cbbc00fb3866c8/spyrit/learning/model_Had_DCAN.py#L742

Besides that, is `evaluate_Pinv()` supposed to call `forward_acquire()` before calling `forward_Pinv()`?

guilhermebene, 2021-07-27:
@nducros `evaluate_Pinv()` is supposed to call `forward_reconstruct_pinv()`, right?
If I am not wrong it should just be:

```
def evaluate_Pinv(self, x):
        with torch.no_grad():
           b,c,h,w = x.shape
           x = self.forward_reconstruct_Pinv(x, b, c, h, w)
        return x
```

nducros, 2021-07-27:
It is bugged, right. 
Actually all the 'no grad' functions should be checked. 
My idea was to have all the `forward_*` functions in a no grad version, without prefix, i..e, 

```
def reconstruct_Pinv(self, x):
        with torch.no_grad():
           b,c,h,w = x.shape
           x = self.forward_reconstruct_Pinv(x, b, c, h, w)
        return x
```

nducros, 2021-09-13:
Solved

closed 2021-09-13 by nducros