openspyrit/spyrit #35: Remove fht

tbaudier, 2021-07-23:


tbaudier, 2021-07-23:
How can I remove the function `fht2` in these files:
https://github.com/openspyrit/spyrit/blob/master/spyrit/misc/pattern_choice.py#L224
https://github.com/openspyrit/spyrit/blob/master/spyrit/learning/model_Had_DCAN.py#L54
https://github.com/openspyrit/spyrit/blob/master/spyrit/reconstruction/basic_hadamard.py#L32

nducros, 2021-07-23:
> https://github.com/openspyrit/spyrit/blob/master/spyrit/misc/pattern_choice.py#L224
keep it as it is for now. Maybe write down somewhere that this function should be updated

> https://github.com/openspyrit/spyrit/blob/master/spyrit/learning/model_Had_DCAN.py#L54
Do not need the function anymore

> https://github.com/openspyrit/spyrit/blob/master/spyrit/reconstruction/basic_hadamard.py#L32
`fht2(X)` --> `wh.walsh2(X,H)/len(X)` with `H = walsh_matrix(len(X))` outside the `for` loop

guilhermebene, 2021-07-30:
@tbaudier We have accepted PR #34, but there are still traces of fht in the `README`, `setup.py`, `.gitmodules`, `.github/workflows/main.yml`, and the symbolic link. Can you take care of this in the future?

closed 2021-09-08 by tbaudier