openspyrit/spyrit #39: Error when creating a DenoiCompNet

guilhermebene, 2021-07-30:
Following the last update where `misc/statistics` was added, I am now getting an error when creating a DenoiCompNet using spas. It seems Cov2Var cannot be found.

```
Traceback (most recent call last):

  File "C:\Users\beneti\spas\scripts\acquisition-recon-sequential.py", line 63, in <module>
    model, device = setup_reconstruction(cov_path, mean_path, H, model_root, network_params)

  File "c:\users\beneti\spas\spas\reconstruction_nn.py", line 146, in setup_reconstruction
    model = DenoiCompNet(

  File "c:\users\beneti\spyrit\spyrit\learning\model_Had_DCAN.py", line 598, in __init__
    super().__init__(n, M, Mean, Cov, variant, N0, sig, H, Ord)

  File "c:\users\beneti\spyrit\spyrit\learning\model_Had_DCAN.py", line 531, in __init__
    super().__init__(n, M, Mean, Cov, variant, H, Ord)

  File "c:\users\beneti\spyrit\spyrit\learning\model_Had_DCAN.py", line 336, in __init__
    Ord = Cov2Var(Cov)

NameError: name 'Cov2Var' is not defined
```

nducros, 2021-07-30:
Right,  `Cov2Var` is now in `misc/statistics`. Can you update master with the import ?

guilhermebene, 2021-07-30:
I am not sure... I might test this in another case rather than in spas to see if I can reproduce the error.

closed 2024-07-10 by romainphan