openspyrit/spyrit #40: Renaming of walsh_ordered

cmouton, 2021-09-06:
Hello,
It seems that some methods related to Hadamard have been renamed, causing the deprecation of some code.

For example, [here](https://github.com/openspyrit/spyrit/blob/20e5c76f84edeb84e566cf26c90b63911fb42781/spyrit/misc/statistics.py#L90) 
and [there](https://github.com/openspyrit/spyrit/blob/ac378472327ba6b1575628088dadaeadbc2cc0a3/spyrit/reconstruction/walsh_ordered_recon.py#L17)

walsh_ordered is called but I could not find a definition of walsh_ordered.

Probably walsh_ordered should be replaced by its new name?

nducros, 2021-09-13:
Sure. You can use `walsh2`, `iwalsh2` or the torch version of it `walsh2_torch`

cmouton, 2021-10-04:
The documentation was missing :
[image]

cmouton, 2021-10-04:
I replaced walsh_ordered by walsh2 and now the following warning disappeared 👍
[image]

cmouton, 2021-10-04:
And the documentation is generated through sphinx 👍 
[image]

cmouton, 2021-10-04:
This issue is solved 👍

closed 2021-10-04 by cmouton