openspyrit/spyrit #134: @romainphan Should we replace the definition of meas2img by that of meas2img2, which seems more general?

nducros, 2024-01-23:
@romainphan Should we replace the definition of meas2img by that of meas2img2, which seems more general?
https://github.com/openspyrit/spyrit/blob/607d0272b5b8548d9c823409e09eda2beb22b2e9/spyrit/misc/sampling.py#L47

_Originally posted by @nducros in https://github.com/openspyrit/spyrit/issues/133#issuecomment-1906293791_

romainphan, 2024-01-23:
@nducros actually the two lines 
y = np.pad(meas, (0, Mat.size - len(meas)))
and 
y = np.pad(meas, ((0, Mat.size - len(meas)), (0, 0)))
produce results with two different shapes. You can check this by checking the bit of code at the end of my comment. From what I understand, the output of the function should have dimensions of (Mx*My , B) and not (Mx*My , Mx*My), hence the second option is preferred. The writing of the second function is also easier to read

###
matrix = np.random.randn(4, 4)

padded1 = np.pad(matrix, ((0, 2)))            # equivalent to meas2img
padded2 = np.pad(matrix, ((0, 2), (0, 0)))   # equivalent to meas2img2

print(padded1.shape)   # outputs (6,6)
print(padded2.shape)    # outputs (6,4)

nducros, 2024-01-23:
Have you tried to run both meas2img and meas2img2 using the same input (a single measurement vector, not a batch).
I am expecting the same results.

romainphan, 2024-01-23:
Yes, in the tutorial 2 you can replace line 135 by
`from spyrit.misc.sampling import meas2img, meas2img2`
and line 138 by
`y_plot = meas2img2(y_plot, Sampling_map)`

This gives an error (related to the shape of the input), but if you still apply the original line [ `y_plot = meas2img(y_plot, Sampling_map)` ] it all works fine.

nducros, 2024-01-24:
OK ! Please can you update the doc of both functions. Consider adding a note or a warning to make the difference explicit.

romainphan, 2024-01-25:
Update done on the dosctring

romainphan, 2024-01-26:
non j'ai fait ça ici https://github.com/openspyrit/spyrit/pull/137

romainphan, 2024-01-30:
Effectivement t'avais raison, meas2img2 est le cas où on a un batch de mesures et on traite tout en même temps, alors que meas2img traite une seule mesure.
ça me fait beaucoup penser à comment PyTorch gère les batchs, mais dans pytorch les batchs sont dans la première dimension (BxNxN par exemple au lieu de NxNxB).

nducros, 2024-01-30:
Yes ! Par contre, je ne sais plus pourquoi j'ai mis les batches à la fin... A terme, on devrait utiliser partout la plus générale des deux fonctions et supprimer l'autre. Tu notes ça dans une issue pour qu'on oublie pas ?

romainphan, 2024-01-30:
Oui on pourrait fusionner les deux fonctions mais j'ai peur que si on en supprime une, toutes les personnes qui utilisent la fonction supprimée vont avoir des problèmes...

closed 2024-01-30 by romainphan

nducros, 2024-01-30:
Oui, c'est un ce que je crains aussi... mais plus on attend plus ce sera critique de fusionner. A méditer