openspyrit/spyrit #143: Spyrit documentation / Separate html page for each module

romainphan, 2024-02-06:
Should we have separate html pages for each module ? The [core package documentation page](https://spyrit.readthedocs.io/en/master/api/spyrit.core.html) is way too long and could be divided into subpages for each package. Same for the [misc package](https://spyrit.readthedocs.io/en/master/api/spyrit.misc.html).
What do you think ?
@tbaudier @nducros

romainphan, 2024-02-08:
@tbaudier @nducros 
J'ai inclus une documentation automatique dans le master, elle fonctionne très bien en local mais échoue sur readthedocs.
La nouvelle branche doc_auto est là pour résoudre le pb

tbaudier, 2024-02-13:
Je n'arrive pas à la faire fonctionner en local et j'ai la même erreur que rtd:

```
python -m venv venv_doc 
source venv_doc/bin/activate
git clone --depth 1 https://github.com/openspyrit/spyrit spyrit 
cd spyrit
git fetch origin --force --prune --prune-tags --depth 50 refs/heads/doc_auto:refs/remotes/origin/doc_auto
git checkout --force origin/doc_auto
git clean -d -f -f
python -m pip install --upgrade --no-cache-dir pip setuptools
python -m pip install --upgrade --no-cache-dir sphinx readthedocs-sphinx-ext
python -m pip install --exists-action=w --no-cache-dir -r requirements.txt
cd docs/source
 python -m sphinx -T -b html -d _build/doctrees -D language=en . $PWD/html
```

J'obtiens la meme erreur:
```
../../spyrit/tutorial/tuto_01_acquisition_operators.py unexpectedly failed to execute correctly:

    Traceback (most recent call last):
      File "/Users/tbaudier/spyrit/doc/spyrit/spyrit/tutorial/tuto_01_acquisition_operators.py", line 54, in <module>
        x, _ = next(iter(dataloader))
      File "/Users/tbaudier/spyrit/doc/venv_doc/lib/python3.9/site-packages/torch/utils/data/dataloader.py", line 631, in __next__
        data = self._next_data()
      File "/Users/tbaudier/spyrit/doc/venv_doc/lib/python3.9/site-packages/torch/utils/data/dataloader.py", line 675, in _next_data
        data = self._dataset_fetcher.fetch(index)  # may raise StopIteration
      File "/Users/tbaudier/spyrit/doc/venv_doc/lib/python3.9/site-packages/torch/utils/data/_utils/fetch.py", line 54, in fetch
        return self.collate_fn(data)
      File "/Users/tbaudier/spyrit/doc/venv_doc/lib/python3.9/site-packages/torch/utils/data/_utils/collate.py", line 277, in default_collate
        return collate(batch, collate_fn_map=default_collate_fn_map)
      File "/Users/tbaudier/spyrit/doc/venv_doc/lib/python3.9/site-packages/torch/utils/data/_utils/collate.py", line 144, in collate
        return [collate(samples, collate_fn_map=collate_fn_map) for samples in transposed]  # Backwards compatibility.
      File "/Users/tbaudier/spyrit/doc/venv_doc/lib/python3.9/site-packages/torch/utils/data/_utils/collate.py", line 144, in <listcomp>
        return [collate(samples, collate_fn_map=collate_fn_map) for samples in transposed]  # Backwards compatibility.
      File "/Users/tbaudier/spyrit/doc/venv_doc/lib/python3.9/site-packages/torch/utils/data/_utils/collate.py", line 152, in collate
        raise TypeError(default_collate_err_msg_format.format(elem_type))
    TypeError: default_collate: batch must contain tensors, numpy arrays, numbers, dicts or lists; found <class 'torchvision.transforms.Compose'>
```

romainphan, 2024-02-15:
J'ai relancé le code que tu as copié ici, et ça marche chez moi en local... ça pourrait être un pb d'OS ?

closed 2024-03-07 by tbaudier