openspyrit/spyrit #155: meas.py / LinearRowSplit example not correct

romainphan, 2024-02-20:
`Example:
        >>> H_pos = np.random.rand(24,64)
        >>> H_neg = np.random.rand(24,64)
        >>> linop = LinearRowSplit(H_pos,H_neg)`

Is this correct ? or should each matri H_pos and H_neg be positive as follows:

`Example:
        >>> H = np.random.rand(24,64)
        >>> H_pos[H > 0] = H[H > 0]
        >>> H_neg[H < 0] = -H[H < 0]
        >>> linop = LinearRowSplit(H_pos,H_neg)`
@tbaudier

nducros, 2024-02-21:
The class can be removed.

The idea will be to merge the [hspim branch](https://github.com/openspyrit/spyrit/tree/hspim) in order to run this [script](https://github.com/openspyrit/spyrit-examples/blob/master-seb/2023_Optica/main_recon_net_EGFP-DsRed_14_all_slices.py)

romainphan, 2024-02-21:
The class LinearRowSplit will be removed.

Concerning the merge, what exactly should be merged ?

nducros, 2024-02-21:
[hspim branch](https://github.com/openspyrit/spyrit/tree/hspim) with `master `such that this [script](https://github.com/openspyrit/spyrit-examples/blob/master-seb/2023_Optica/main_recon_net_EGFP-DsRed_14_all_slices.py) runs

closed 2024-10-01 by nducros