openspyrit/spyrit #176: double definition of "Permutation_Matrix", in spyrit.misc.matrix_tools and spyrit.misc.sampling

romainphan, 2024-03-01:
### Issue
Double definition of the same function, in `spyrit.misc.matrix_tools` and in `spyrit.misc.sampling`. Both definitions are identical, the first one doesn't seem to be used in any other file.

### Suggestion
Should we eventually remove `spyrit.misc.matrix_tools.Permutation_Matrix` ? 
I think we should first link `spyrit.misc.matrix_tools.Permutation_Matrix` to `spyrit.misc.sampling.Permutation_Matrix` and print a user warning, then in a few months the function could be effectively removed.

nducros, 2024-03-26:
`Permutation_Matrix` still in use HadamSplit (and elsewhere?)

Idea :  Remove `Perm` everywhere, keep the corresponding indices as an nn.parameters.

romainphan, 2024-03-26:
The function in matrix_tools has been linked to the one in sampling.

closed 2024-03-26 by romainphan