openspyrit/spyrit #196: Documentation source files are versionned

nducros, 2024-05-17:
Some folders have been included by mistake 
* [_autosummary](https://github.com/openspyrit/spyrit/tree/master/docs/source/_autosummary)
* [_static/css](https://github.com/openspyrit/spyrit/tree/master/docs/source/_static/css)
* [_templates](https://github.com/openspyrit/spyrit/tree/master/docs/source/_templates)
* [api/generated/backreferences](https://github.com/openspyrit/spyrit/tree/master/docs/source/api/generated/backreferences)
* [html](https://github.com/openspyrit/spyrit/tree/master/docs/source/html)
* ... ?

@romainphan: Please discuss it with Thomas, as removing the files does not remove them from the history.

nducros, 2024-05-17:
I removed them from version control *not* from history and included some folders to the .gitignore file.

tbaudier, 2024-05-17:
I did a branch :

I removed that folder from the history and put them in .gitignore. The docs is here:

romainphan, 2024-05-23:
everything correct now ? 

_templates has to be kept (it determines the .rst files for the automatic documentation)
_static/css could have been kept (it determines the css for the tutorial homepage - but it is optional)
the others have been correctly deleted

nducros, 2024-05-23:
I am not sure about _templates. You can check the [doc from my branch](https://spyrit.readthedocs.io/en/clean_tuto_cdnet_drunet/_autosummary/spyrit.core.meas.DynamicLinear.html). I deleted _templates but I cannot see what's missing.

nducros, 2024-05-24:
I agree for _templates :) 
What about \docs\source\sg_execution_times.rst

closed 2024-05-25 by nducros