openspyrit/spyrit #207: sampling masks

nducros, 2024-05-25:
J'ai fusionné les différentes branches et tout me semble OK. 
J'ai essayé de créer les "naive" et "variance" sampling masks de la même façon 
```
idx = np.argsort(Ord_nai.ravel(), axis=None)[::-1]
mask_nai = np.zeros_like(Ord_nai)
mask_nai.flat[idx[0:M]] = 1
```
mais j'obtiens une figure bizarre, ce que tu solutionne avec
```
mask_nai = np.zeros((h, h))
mask_nai[0 : int(M / h), :] = 1
```

romainphan, 2024-05-27:
Le problème vient du np.argsort(), qui utilise par défaut un quicksort, qui ne conserve pas l'ordre quand il y a plusieurs valeurs égales. Il faut donc écrire :

`idx = np.argsort(-Ord_nai.ravel(), axis=None, kind='stable')
mask_nai = np.zeros_like(Ord_nai)
mask_nai.flat[idx[0:M]] = 1`

à noter la différence qu'on doit écrire `-Ord_nai.ravel()` contrairement à `Ord_nai.ravel() ... [::-1]`. Pour le "variance subsampling" ça devient donc :
`idx = np.argsort(-Ord_var.ravel(), axis=None, kind='stable')`

romainphan, 2024-05-27:
Hehe c'est déjà fait, c'est ma fonction sort_by_significance
J'ai modifié tout ça dans ma dernière version sur ma branche

romainphan, 2024-05-27:
Tout fonctionne, à fusionner sur le master

closed 2024-05-27 by romainphan