openspyrit/spyrit #219: Denoising layer of DC Net

nducros, 2024-07-03:
https://spyrit.readthedocs.io/en/master/_autosummary/spyrit.core.recon.Denoise_layer.html#spyrit.core.recon.Denoise_layer
* Equation does not display well
* Notations not consistent (\sigma vs. A=)

romainphan, 2024-07-04:
corrected

https://github.com/openspyrit/spyrit/pull/221/commits/6628dee8403a3ff04dfa767bd9e2c99d82e31343

nducros, 2024-07-04:
The doc is confusing as the denoise layer only compute the filter, which is designed to be multiplied to some input measurement vector. The usage is only visible in here: https://github.com/openspyrit/spyrit/blob/aa1739794e206e9441ccc228a0b77a2be6e087f9/spyrit/core/recon.py#L191

The design of the class seems awkward, as the forward function should actually compute what the doc says, i.e., (a / a + b) input

romainphan, 2024-07-16:
https://github.com/openspyrit/spyrit/issues/228

romainphan, 2024-10-21:
in https://github.com/openspyrit/spyrit/pull/244/commits/80f664b31b00f30964c142f04601aef455da0f96
Denoise_layer is removed ; TikhonovMeasurementPriorDiag no longer uses Denoise_layer but an internal method, much more simple to understand

closed 2024-11-05 by romainphan