openspyrit/spyrit #220: LearnedPGD

nducros, 2024-07-03:
See [here](https://spyrit.readthedocs.io/en/master/_autosummary/spyrit.core.recon.LearnedPGD.html#spyrit.core.recon.LearnedPGD)

* Example are copied from Pinv (not up to date)
* @jabascal Do you confirm we can remove `reconstruct_expe`?

nducros, 2024-07-04:
Yes. Most methods of the class have the same errors.

jabascal, 2024-07-04:
>    - Examples are copied from Pinv (not up to date)

@ducros

Not needed for new classes created. At fig_recon_sim_prelim.py, I am doing

                    if net_arch == 'lpgd' or net_arch == 'dfb-net':
                        rec_gpu = model.reconstruct(m, exp=True)
                    else:
                        model.prep.set_expe()
                        rec_gpu = model.reconstruct_expe(m)
                    rec = rec_gpu.cpu().detach().numpy().squeeze()

closed 2024-10-01 by nducros