openspyrit/spyrit #234: How can I change the code in the function package

WH-CSU, 2024-09-12:
Dear Author, How can I change the code in the function package? For example, spyrit.core.noise.Poisson, I found only comments and Example in the noise.py function package. thanks if you can reply!

romainphan, 2024-09-12:
Hi, I assume you want to change the source code to implement yourself some changes, for instance in noise.py ?
If that's the case, you can follow the guidelines to install spyrit in developer mode [in the ReadMe](https://github.com/openspyrit/spyrit). It works for Windows too even if it's in the Linux & MacOs section.
I hope this helps,

nducros, 2024-10-01:
Hi,
Have you checked it works?

WH-CSU, 2024-10-04:
> Hi, Have you checked it works?

Thank you very much for your help, I made changes to the code, although I'm not sure what the difference is between developer mode and application mode, but changing the noise.py code in application mode still works!

romainphan, 2024-10-07:
Hi, glad to know this works :) The developer mode is useful if you want to have your package files somewhere else than in the (default) environment directory. This way you can have multiple environments referring to the same modified version.
By the way, you're more than welcome if you'd like to contribute to the package!

WH-CSU, 2024-10-10:
> Hi, glad to know this works :) The developer mode is useful if you want to have your package files somewhere else than in the (default) environment directory. This way you can have multiple [...]

Thank you very much for your help, and once the results of my work prove useful, I'll be happy to make the modified code public so that more people can learn about spyrit!