openspyrit/spyrit #240: Experimental results input

WH-CSU, 2024-10-13:
Dear authors, Is the use of this program for processing the results of an experiment reading in the y-values? (Sampling order is consistent with simulation)
Is there a program to read the y-value of an experiment and denoise it using this network?

romainphan, 2024-10-14:
Hello, in [spyrit.core.recon](https://spyrit.readthedocs.io/en/master/_autosummary/spyrit.core.recon.html) you can find several classes (PinvNet being the most simple one, DCNet and LearnedPGD) that contain a `reconstruct_expe` method that specifically reconstructs images from measurements.

romainphan, 2024-10-17:
Hi again, did this help you ?

romainphan, 2024-11-05:
I'm closing this issue, but feel free to reopen it or another one if you need any help.

closed 2024-11-05 by romainphan