openspyrit/spyrit #272: Issue when importing spyrit with pip

Tommte, 2025-01-15:
There is a problem when installing the package via pip. After creating a new conda env and using ``` pip install spyrit ``` , the following error shows : 
[image]

romainphan, 2025-01-15:
No idea what causes this error.... try specifying a version when installing ?
Maybe @tbaudier you have an idea ?

Tommte, 2025-01-15:
Error was due to a new version of Python for which at least torchvision had no compatible version

Tommte, 2025-01-16:
Some precisions : torchvision is incompatible with Python 3.13

[image]

Apparently install might work also when using conda to install torch before installing spyrit because it switch to Python 3.12

tbaudier, 2025-01-27:
Hello,

Yes you need to create a conda environment with a python version < 3.13 and then you can install your packages with pip

closed 2025-01-27 by tbaudier