openspyrit/spyrit #282: HadamSplit2d: pinv method with if self.fast as for adjoint and measure?

romainphan, 2025-02-06:


romainphan, 2025-02-06:
There is already a `fast_pinv` method. We have already discussed this in length ("Should pinv be in the measurement class ?") and it seemed to me pinv should not be a method of the measurement operator.

However, if you still wish to iimplement it i can do it.

nducros, 2025-02-07:
1. We do have a pinv method in `HadamSplit2d`: it's called `fast_pinv`. This is why I would rename `fast_pinv` to `pinv`.
2. I see no point in keeping the `fast` attribute, but tell me. The user can call parent's methods from an `HadamSplit2d` operator. E.g.:

"Fast  measure"
```
import torch
import spyrit.core.meas as meas
h = 32
meas_op = meas.HadamSplit2d(h, 49)
x = torch.empty(8, 2, h, h).uniform_(0, 1)
y = meas_op.measure(x)
```

"Slow measure from LinearSplit"
```
y = super(type(meas_op), meas_op).measure(x)
or 
y = LinearSplit.measure(meas_op, x)
```